booya-at/OpenGlider #30: Unwrap failure

Mx74, 2018-04-13:
Not sure if this issue is due to freecar_glider or openglide, but as most of the error are about Openglider I post it here:

with the last commits for both freecad_glider and openglider, the unwrap glider glider fails with this error:

`Running the Python command 'PatternCommand' failed:
Traceback (most recent call last):
  File "/usr/lib/freecad-daily/Mod/freecad/freecad_glider/tools/__init__.py", line 179, in Activated
    pat.unwrap(file_name, obj.Proxy.getGliderInstance())
  File "/home/max/OpenGlider/openglider/plots/__init__.py", line 73, in unwrap
    plots.unwrap()
  File "/home/max/OpenGlider/openglider/plots/glider/__init__.py", line 169, in unwrap
    self.get_dribs()
  File "/home/max/OpenGlider/openglider/plots/glider/__init__.py", line 104, in get_dribs
    dribs = self._get_cellplotmaker(cell).get_dribs()
  File "/home/max/OpenGlider/openglider/plots/glider/cell.py", line 551, in get_dribs
    dribs.append(drib_plot.flatten(self.attachment_points))
  File "/home/max/OpenGlider/openglider/plots/glider/cell.py", line 432, in flatten
    return self._flatten(attachment_points, self.config.drib_num_folds)
  File "/home/max/OpenGlider/openglider/plots/glider/cell.py", line 471, in _flatten
    self._insert_attachment_points(plotpart, attachment_points)
  File "/home/max/OpenGlider/openglider/plots/glider/cell.py", line 407, in _insert_attachment_points
    p1, p2 = self.get_left(attachment_point.rib_pos)
  File "/home/max/OpenGlider/openglider/plots/glider/cell.py", line 334, in get_left
    return self.get_p1_p2(x, side=0)
  File "/home/max/OpenGlider/openglider/plots/glider/cell.py", line 381, in get_p1_p2
    assert front[0] <= x <= back[0]
`

looooo, 2018-04-13:
sry, this is still work in progress. I tested to unwrap the default-glider and it fails with a similar issue:

```
Traceback (most recent call last):
  File "/home/kati/projects/freecad_glider/freecad/freecad_glider/tools/__init__.py", line 179, in Activated
    pat.unwrap(file_name, obj.Proxy.getGliderInstance())
  File "/home/kati/projects/OpenGlider/openglider/plots/__init__.py", line 87, in unwrap
    all_patterns.export_dxf(fn("plots_all_dxf2000.dxf"))
  File "/home/kati/projects/OpenGlider/openglider/plots/drawing/layout.py", line 503, in export_dxf
    group = drawing.groups.new()

'Drawing' object has no attribute 'groups'
```
I guess @hiaselhans knows how to solve this ; )

Mx74, 2018-04-16:
Oh, ok. 
But it was working with an earlier commit, that's why I opened this issue.

looooo, 2018-04-18:
@Mx74 can you send the problematic glider? (json or freecad-file)

Mx74, 2018-04-19:
Sure:
[test.zip](https://github.com/booya-at/OpenGlider/files/1927715/test.zip)

looooo, 2018-04-19:
thx. I guess there is a problem with the diagonal input. The order is sometimes not correct.

[image]