booya-at/OpenGlider #32: Workbench launch failure

Mx74, 2018-07-11:
I was wondering how was running the install process since you integrated freecad-glider with OpenGlider. 

Everything ran smooth without warning, the workbench icon appears in Freecad, but when trying to launch it:

`('invalid syntax', ('/home/max/OpenGlider/openglider/mesh/poly_tri.py', 401, 22, '            return [[*edge] for edge in zip(_tri[:-1], _tri[1:])]\n'))
Traceback (most recent call last):
  File "/home/max/OpenGlider/freecad/freecad_glider/init_gui.py", line 101, in Initialize
    from . import tools
  File "/home/max/OpenGlider/freecad/freecad_glider/tools/__init__.py", line 15, in <module>
    from . import _glider as glider
  File "/home/max/OpenGlider/freecad/freecad_glider/tools/_glider.py", line 8, in <module>
    from openglider import mesh
  File "/home/max/OpenGlider/openglider/mesh/__init__.py", line 1, in <module>
    from openglider.mesh.mesh import Mesh, Vertex, Polygon
  File "/home/max/OpenGlider/openglider/mesh/mesh.py", line 12, in <module>
    from .poly_tri import PolyTri
`

looooo, 2018-07-11:
can you try import PolyTri from python:
`from openglider.mesh.poly_tri import PolyTri`

Mx74, 2018-07-11:
!!!!NOT ABLE TO IMPORT MESHPY, MESH-CREATION WILL BE SLOW
No module named meshpy.triangle
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/max/OpenGlider/openglider/mesh/__init__.py", line 1, in <module>
    from openglider.mesh.mesh import Mesh, Vertex, Polygon
  File "/home/max/OpenGlider/openglider/mesh/mesh.py", line 12, in <module>
    from .poly_tri import PolyTri
  File "/home/max/OpenGlider/openglider/mesh/poly_tri.py", line 401
    return [[*edge] for edge in zip(_tri[:-1], _tri[1:])]
             ^
SyntaxError: invalid syntax

looooo, 2018-07-11:
poly_tri didn't work with python2. This should be fixed: https://github.com/booya-at/OpenGlider/commit/9614ff85dea51e4a6fda37850c5d1689abcf01ca

But the problem is meshpy: It seems like you have removed the meshpy package.

Mx74, 2018-07-11:
I haven't removed anything as I tried on a new virtual machine with a fresh install to see if the install process was less painful. It's just not installed. I'll try to fix this

looooo, 2018-07-11:
I see. Btw. we are using conda with virtual environments which allows to have multiple environments side by side...
further information: https://github.com/FreeCAD/FreeCAD_Conda

Mx74, 2018-07-12:
Ok.

I see that there is a conda recipe for Openglider. 

I tried conda install -c openglider openglider , install run fine, but then how do you get the workbench active in Freecad?

looooo, 2018-07-12:
for conda you have to use freecad from conda-forge:

`conda create -n freecad freecad openglider meshpy -c conda-forge -c openglider`
this creates an environment with all the needed tools. Note that this environment is quite huge.

once everything installed you can activate the environment with:
`source activate freecad`

and launch freecad:
`FreeCAD`

if you want the latest development-state of openglider use pip in the environment:
`pip install https://github.com/booya-at/OpenGlider/archive/master.tar.gz`

Mx74, 2018-07-13:
Hello,

Ok, it works with conda, the install process is indeed much easier :-)

There is just the file glider2d.json missing in the package at:
/miniconda3/envs/freecad/lib/python3.6/site-packages/freecad/freecad_glider/

looooo, 2018-07-13:
thanks for reporting. It is already fixed in the development-branch.

Mx74, 2018-07-13:
For the install of the dev branch, I tried with pip, it failed with mistake. Sorry, I didn't note which one. 
Then I cloned it with git, and installed the usual way. There were cython, lxml and gmsh_interop dependencies missing. 
Once installed, it seems it works. 

Thank you for your help !