booya-at/OpenGlider #35: Usage instructions not thorough enough

tuskiomi, 2018-10-26:
The current Instructions on installing and using the package are very incomplete. Please add instructions on using the packages after installation.

Iv, 2018-12-17:
Simple demo project would be much appreciated!

Mx74, 2018-12-17:
Actually there is one demo included. 
When you launch the workbench, create a new file in Freecad, and then press the first icon (the small paraglider).
It creates a mini wing, with everything, allowing you to understand how the software works. Canopy, ballooning, design cut, lines, air intake, cross rib, etc...

To see it better, in the properties windows, view tab, you can change the value of hole_num, line_num, etc...

You can use it as a base in order to create your own glider. 

I'm still not mastering it fully, but I may try to make some screecast showing the basic if Loooo agrees.

looooo, 2019-01-06:
> I'm still not mastering it fully, but I may try to make some screecast showing the basic if Loooo agrees.

@Mx74  any contribution is welcome. Maybe best to add docs to the github-wiki.