booya-at/OpenGlider #47: Generate the unwrapped pattern as a sketcher object in FreeCAD instead of straight to DXF file

goatchurchprime, 2020-07-18:
It would be more versatile if the cut pattern were converted into FreeCAD Sketcher instead of being dumped into a several DXF file that can only be seen using another system, like LibreCAD.  Then it would be available for further processing.

I was able to load it back into FC using my DXF_to_Sketch_layers macro.  The relevant function to use is [here](https://github.com/goatchurchprime/transition-CAM/blob/master/dxfimporting/Macro_DXF_to_Sketch_layers.FCMacro#L110).
[image]

If the cut pattern were in FreeCAD I'll will have more reasons to finish porting across features from my javascript-web-based [laserplacer](https://goatchurch.bitbucket.io/laserplacer.html) into the Path workbench so that it can output the [output codes](https://bitbucket.org/goatchurch/laserplacer/src/master/pencutseqtocode.js) used by the laser cutter.  

The Laserplacer is actually used by @trilbytim to place and cut sails at Avian Hang Gliders from DXF designs in LibreCAD, but it should be part of FreeCAD instead of being a stand-alone unfinished program that will eventually be too out of date to use.  The laser cutter he uses has a 15m long bed, draws pen lines as well as cuts, and is the same one as seen on videos of paraglider manufacturing.  

It was actually able to load an OpenGlider output file, but the layers and colours are not in the form expected, so I couldn't test whether the algorithms could group the cuts and pen-markings into their proper closed contour pieces.
[image]

Last time I worked on this I met some significant challenges with FreeCAD development (eg not being able to pip install ezdxf). Maybe someone here could advise me about work-arounds and outline how best to approach the steps I need to solve in order to produce something that can be used more generally for any laser cutting layout.  I would prefer to spend my time working on the contour grouping algorithms instead of trying to beat the FreeCAD system in ways it cannot be beat!

hiaselhans, 2020-07-18:
hi @goatchurchprime,

if you feel like it, the plots created are actually a [Drawing](https://github.com/booya-at/OpenGlider/tree/develop/openglider/vector/drawing) instance which could very likely be converted directly to a FreeCAD sketch.
However, the patterns are made as they were required by a-name factories (cuts in red, marks in green, seperate layers and all elements grouped).
That should do for common nesting software and commercial CAD. Actually most of your weird and long layer names must emerge somewhere else, here is the dxf imported to some "quite known" commercial CAD:
[image]

I am not sure what is the final goal you are trying to achieve, but if you come up with a PR to load the "Drawing" back in the sketcher i think it is highly appreciated by @looooo  who knows better about the FreeCAD internals and politics.
We initially decided to output dxf directly because it is the most "portable" format.

goatchurchprime, 2020-07-19:
There must be a bug in my laserplacer code in the way it additionally breaks up layers by colours. After editing the file in LibreCAD I got the groupings to work and place the pieces onto a stock material. 
[image]

---->  Do you know what is the software used in the factory to group the cut elements into closed contours and assign the correct pen-markings to them before repositioning them as groups in the nesting process?  I have not seen where this is done, and I would like to review its GUI.  <-----

Usually when I watch people prepare files for laser cutting they painstakingly select lines using only rectangle select and individual item select before dragging the sets of pieces where they want.  Their software does not understand groups defined by closed connected geometry.  

The goal I would like to achieve is for the final stages in the process -- (a) selecting and allocating contours to different sheets of material, (b) nesting the contours, and (c) outputting the .anc or .hpgl file to the laser cutter -- to be completable in FreeCAD.  

As I said, I have implemented this process once in Javascript, and it is in use at Avian Hang-gliders (that's a paraglider connection!).  For the hang-glider wings we do the nesting by hand because there are fewer pieces and they only just fit on the stock roll (by design), but I have made an experiment using SVGNest from FreeCAD here: https://forum.freecadweb.org/viewtopic.php?t=33485

If this final goal worked, there are obviously a lot of other applications for running laser cutters than only making wings.  Some of the architecture applications generate hundreds of shapes to be laser cut from plywood.  

But the first change would be to break your workbench feature into two components: (a) create sketcher with layers, (b) output sketcher to dxf, with everything that uses that dxf still working.  That would prove that all the information is there and provide something more than a toy example to work against.

looooo, 2020-07-20:
integrating the patterns into freecad makes sense. But therefore it's necessary to understand the drawing stuff. btw.: I don't think sketches are the right structure. Better to use more basic wires to represent the patterns in freecad.