booya-at/OpenGlider #51: Glider workbench appearing as default in Freecad 0.19dev

juliencarnot, 2020-10-24:
Not sure if this is the right place, but I was surprised to see this very cool but very specific workbench in the default and active  workbenches of my Freecad 0.19dev appimage install. 

It doesn't appear in the addons manager, and can't be deactivated.
Please close this if I'm the only one or if you can't do anything about it, and keep up the good work!

looooo, 2020-12-14:
Sorry for this inconvenience. There are several issues with providing this workbench via the addon-manager. I tried several things but was not able to provide an easy workflow. The workbench is included only temporarily and will not be part of the release.

looooo, 2021-04-18:
The glider workbench is not part of the 0.19 release bundles but still included in the 0.20-weekly builds.

closed 2021-04-18 by looooo