booya-at/OpenGlider #62: Exporting Mesh on Windows using miniconda

aureliencousin, 2021-11-14:
Hi,

I am struggling exporting a paraglider mesh. I've tried dae, obj, dxf, stl but everytime it looks like some lib required by MeshPart is missing. This is the error I'm getting:

>   File "<string>", line 3, in <module>
>   File "C:\Users\Acacia\miniconda3\envs\openglider\Library\Mod\Arch\importOBJ.py", line 22, in <module>
>     import FreeCAD, DraftGeomUtils, Part, Draft, Arch, Mesh, MeshPart, os, sys
> <class 'ImportError'>: DLL load failed while importing MeshPart: La procédure spécifiée est introuvable.

Setup is FreeCAD 0.19.2 on Windows 10 using miniconda3 with python 3.7.
I installed openglider using miniconda by following the instructions described here: https://booya-at.github.io/openglider-tutorial/00_installation.html

Also, I read that at some point there were FreeCAD bundles with the glider-workbench preinstalled, however I can't find any working download link for any of those and the few versions I've tried don't include the workbench. Could you provide a download link for one of those ? I would like to try if this fixes the issue.

Thanks, your work is amazing !

aureliencousin, 2021-11-14:
I tried it on ubuntu 20.04 and got the same issue

> 17:31:05  Traceback (most recent call last):
>   File "<string>", line 3, in <module>
>   File "/home/aurelien/miniconda3/envs/openglider/Mod/Arch/importDAE.py", line 22, in <module>
>     import FreeCAD, Mesh, os, numpy, MeshPart, Arch, Draft
> <class 'ImportError'>: /home/aurelien/miniconda3/envs/openglider/lib/MeshPart.so: undefined symbol: _ZN9SMESH_Gen7ComputeER10SMESH_MeshRK12TopoDS_Shapei13MeshDimensionPSt3setIiSt4lessIiESaIiEE

Is there a working way to export the 3D object for later use on some other CAD software or for 3D printing ?

hiaselhans, 2021-11-17:
hi @aureliencousin 

this seems to be a freecad compilation issue.
@looooo any idea?

looooo, 2021-11-17:
this is related to smesh. I guess it's because of wrong pinning. Please try to install latest freecad with `conda install freecad -c freecad/label/dev` or install smesh with `conda install smesh=9.6.*`

aureliencousin, 2021-11-21:
Thanks for your help  @looooo @hiaselhans. Installing the latest version fixes the MeshPart lib not found error. 
However, I'm always getting an empty mesh when exporting the glider

```
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a----        21/11/2021     12:19            448 Unnamed-Glider.dae
-a----        21/11/2021     12:38             49 Unnamed-Glider.obj
-a----        21/11/2021     12:37             84 Unnamed-Glider.stl
```

Does OpenGlider support export as a 3D Mesh ?

looooo, 2021-11-22:
you need a format which export the viewprovider-mesh. `wrl` works fine for me.

aureliencousin, 2021-11-30:
Thanks @looooo , wrl export seems to work (but unfortunately, I can't import my glider in blender as there are some parsing issues).

I took some time to read the OpenGlider code, it looks like at some point it was possible to export the mesh as obj but that this feature is now deprecated. I think it would be great if an export to a standard format such as .obj was possible. Not much of a python dev myself but I might contribute when I have some time.

Thanks for your help, I guess you can close the issue as the initial problem was solved after upgrading freecad to the latest dev tag