booya-at/OpenGlider #66: Total line length

Mx74, 2021-12-07:
Hello,

For kite design, that would be convenient to be able to put the pilot point further down than 10m. 
Is there a technical reason for limiting this distance to 10m? If not, removing the limit would be a plus.
Mx

looooo, 2021-12-15:
thanks for reporting:
https://github.com/booya-at/OpenGlider/commit/1ea23e47afde56b63dd8ba4769644a43a915db2c

Mx74, 2021-12-16:
Yep, the 10m limitation is gone.
It somehow works; but adding a fifth level of line (high, mid, low, mixer + long lines) mess up the line calculation.

looooo, 2021-12-16:
normally there shouldn't be a limitation. can you send an example?

Mx74, 2021-12-16:
Indeed, this is not only a number of level problem, but of of lenght.
If I set my mixer too short, then some weird things happen.
WIth 4 levels, setting the risers at 40cm isn't a problem. 

But with 5 levels, this is what happens: 
Mixer (equivalent of risers) at 1m or 90cm, everything is fine
[image]

Mixer at 30cm or 50cm: weird angle
[image]

looooo, 2021-12-16:
oh I see this is a problem of overcompensating the residual forces. You can try to set the factor at the end (0.5) to a smaller value to reduce the compensation per step. Some more explanation is inside the code.
https://github.com/booya-at/OpenGlider/blob/develop/openglider/lines/lineset.py#L330L334

Mx74, 2021-12-17:
Perfect, thanx a lot. 
Made a simple trial with 0.3, and it looks better. 
I'll investigate more for finding the right parameter.

Mx74, 2021-12-17:
Just a thought, as it's hard coded, this change will disappear at every update. Not a big deal when you know, but for people not aware of it, would it be possible to integrate it as a parameter in the gui?

looooo, 2021-12-17:
It's not the best solution anyway. I guess we can store a value for every line and look if the residual gets smaller for every run. If the residual gets bigger choose a smaller value.
But I guess there is also the option to minimize the whole system at once.

Mx74, 2021-12-17:
Yep, and just changing the parameter isn't as easy as it looks. Indeed, I did a rough test without really taking care of the final result (you can see really long upper lines for example), but then trying to refine the model to get coherent lenght for the upper lines seems to be complex.

looooo, 2021-12-17:
what do you mean with `coherent`?

Mx74, 2021-12-17:
realistic?

looooo, 2021-12-17:
ah ok, yes you are right the process is quite time-intensive, I guess changing the line-length on the 3d model would be more sufficient.

Mx74, 2021-12-21:
Yep, it would just have been convenient to ge tthe total line ength, in order to avoid guessing the space between the 2 sides.
But that's easy to work around. 
Thanks

closed 2021-12-21 by Mx74