booya-at/OpenGlider #67: Scale feature: lock the width of attachment points

Mx74, 2021-12-29:
Hello,

The scale glider function seems to make a perfect full homothetic. Including the width of the attachment points. (harness chest belt length for paraglider). But this specific lenght isn't homothetic (I think that's the only one). 

It would be great if when scaling, we could at least keep it fixed, or better, parametrize this one (second parameter for the two main points in the create/modify line function)

looooo, 2022-01-16:
haven't tested, but this should fix the issue:
https://github.com/booya-at/OpenGlider/commit/eb29d6be170fcf4c5d8c9adb7a3773b0fc4c6aaa

Mx74, 2022-01-21:
It kinda works without naming the riser, except that the riser itself is scaled, which isn't usually the case: risers have a constant length whatever the size.

[image]

When assiging the line type "riser" it does the same.

But when giving a name including "riser", scaling fails with the following errors:

```
1450609:05:23  Traceback (most recent call last):
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/freecad/freecad_glider/tools/glider.py", line 483, in onChanged
    self._updateData(vp, prop)
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/freecad/freecad_glider/tools/glider.py", line 418, in _updateData
    if not self.getGliderInstance():
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/freecad/freecad_glider/tools/glider.py", line 378, in getGliderInstance
    return self.obj.Proxy.getGliderInstance()
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/freecad/freecad_glider/tools/features.py", line 337, in getGliderInstance
    glider.scale(self.obj.scale)
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/openglider/glider/glider.py", line 297, in scale
    self.lineset.scale(faktor)
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/openglider/lines/lineset.py", line 74, in scale
    self.recalc()
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/openglider/lines/lineset.py", line 189, in recalc
    self._calc_geo()
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/openglider/lines/lineset.py", line 205, in _calc_geo
    tangential = self.get_tangential_comp(line, lower_point)
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/openglider/lines/lineset.py", line 326, in get_tangential_comp
    r = self.get_residual_force(line.upper_node)
  File "anaconda2/envs/freecad/lib/python3.9/site-packages/openglider/lines/lineset.py", line 625, in get_residual_force
    residual_force += line.force * line.diff_vector
<class 'TypeError'>: unsupported operand type(s) for *: 'NoneType' and 'float'
```

looooo, 2022-01-21:
should work with https://github.com/booya-at/OpenGlider/commit/2324fb856de979d0d2e9343e00040d3bd47e8b37

Mx74, 2022-01-21:
Well done !

closed 2022-01-21 by Mx74