booya-at/OpenGlider #68: Unwrap error

Mx74, 2022-01-24:
Unwrapping was working so far, but since the 2 last updates: 

float() argument must be a string or a number, not 'NoneType'Running the Python command 'PatternCommand' failed:
Traceback (most recent call last):
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/freecad/freecad_glider/tools/__init__.py", line 167, in Activated
    pat.unwrap(file_name, obj.Proxy.getGliderInstance())
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/openglider/plots/patterns.py", line 157, in unwrap
    super().unwrap(outdir)
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/openglider/plots/patterns.py", line 112, in unwrap
    all_patterns = self._get_plotfile()
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/openglider/plots/patterns.py", line 89, in _get_plotfile
    plots.unwrap()
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/openglider/plots/glider/__init__.py", line 164, in unwrap
    self.get_ribs()
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/openglider/plots/glider/__init__.py", line 84, in get_ribs
    rib_plot.flatten(self.glider_3d)
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/openglider/plots/glider/ribs.py", line 228, in flatten
    return super(SingleSkinRibPlot, self).flatten(glider)
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/openglider/plots/glider/ribs.py", line 83, in flatten
    self.draw_rib(glider)
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/openglider/plots/glider/ribs.py", line 250, in draw_rib
    single_skin_cut = self.rib.profile_2d(singleskin_cut_left)
  File "/miniconda3/envs/freecad/lib/python3.9/site-packages/openglider/airfoil/profile_2d.py", line 49, in __call__
    xval = float(xval)

float() argument must be a string or a number, not 'NoneType'`
`

Mx74, 2022-01-28:
I made a simple test:

- new doc, I create a new glider by default: unwraping works
- I name a line or a riser : unwraping doesn't work anymore. 
- I delete the name, unwraping still doesn't work. 
So it seems it has something to deal with naming those lines

looooo, 2022-02-02:
I can't reproduce on linux/ubuntu.

looooo, 2022-02-02:
btw. please use ` ```code``` ` for code blocks.

Mx74, 2022-02-02:
I'm also on Ubuntu. 
Made another test on Manjaro, same problem. 
Both are installed via Conda.

Mx74, 2022-02-02:
These two different errors made me wonder...

I just copied all the openglider's files again over the existing ones, and now it runs smoothly. 
Don't know what has happen when I updated the last commit, but obviously something went wrong. 
Sorry for the hassle.

closed 2022-02-02 by Mx74

j2l, 2022-03-23:
I just had the exact same error, after I modify the default glider's shape in my case.
To be clear, I add a glider, unwrap glider, it works. Then I modify it and try to unwrap and it fails.
At the end, it throws `Cannot normalize a vector of length 0.0`

On Ubuntu 21.10,  FreeCAD FreeCAD_0.19.3-Linux-Conda_glibc2.12-x86_64.AppImage